fquinto/bticinoClasse300x #1: Support BIT-3488

ubettoni, 2022-07-26:
Thank you for your great Job! I'm asking if you think is possible to make the same hack to the Hometouch BTI-3488.
Thank you!

bubez81, 2022-07-26:
I downloaded the firmware and discovered the password (SMARTDES) but I still haven't had the courage to try

bubez81, 2022-07-28:
hello, @fquinto I need your help :)
The structure of the firmware for the Hometouch 3488 is similar to the 300x:
[image]
[image]
[image]

I managed to mount the btweb_only.ext4 image and make all the necessary changes, but from here I can't recreate the firmware file so that it can be installed on the video door entry unit

this is the link of the firmware https://www.homesystems-legrandgroup.com/home/-/productsheets/2641729
and SMARTDES is the password

Can you help me?

fquinto, 2022-08-10:
Yes, you can see a new tool on this repo (see main.py file)
Running vídeo here: https://asciinema.org/a/514007

fquinto, 2022-08-10:
> Thank you for your great Job! I'm asking if you think is possible to make the same hack to the Hometouch BTI-3488. Thank you!

Yes, you can see a new tool on this repo (see main.py file)
Running vídeo here: https://asciinema.org/a/514007

bubez81, 2022-08-10:
Hi! Great work but is not compatible with the firmware I have posted, the structure of the firmware is similar but not the same.
Did you try to open the SMARTDES firmware to see the structure inside? I am able to find the right files and to do the changes but can't repack the firmware correctly and the flashing tool give me an error (not with your tool but only by interpreting the commands of the how to)
With the new tool I tried to change something like the url and the passwords:

`self.filename = 'SMARTDES_020508.fwz'`
`self.url = f'https://www.homesystems-legrandgroup.com/MatrixENG/liferay/bt_mxLiferayCheckout.jsp?fileFormat=generic&fileName={self.filename}&fileId=58107.23188.18516.42630'`
and
`self.password = 'SMARTDES'`

but obviously it gives me error:

`UnGZ firmware... Traceback (most recent call last):
  File "main.py", line 277, in <module>
    c.main()
  File "main.py", line 44, in main
    self.unGZfirmware()
  File "main.py", line 121, in unGZfirmware
    with gzip.open(f'{self.workingdir}/{self.partFirmware}', 'rb') as f_in:
  File "/usr/lib/python3.8/gzip.py", line 58, in open
    binary_file = GzipFile(filename, gz_mode, compresslevel)
  File "/usr/lib/python3.8/gzip.py", line 173, in __init__
    fileobj = self.myfileobj = builtins.open(filename, mode or 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/bticino-onokbp7t/btweb_only.ext4.gz'
`

because gof the different structure inside

fquinto, 2022-08-11:
Test the same with version 0.0.2.
But I think this firmware it's signed using SHA1    :/

bubez81, 2022-08-11:
hello, for the moment I received this error:
`reating temporary folder... created /tmp/bticino-h40qo0a9 ✅
Downloading firmware... 
-1 / unknown downloaded SMARTDES_020508.fwz ✅
Reading files inside firmware... done ✅
Selecting firmware file... important file is btweb_only.ext4.gz.sha1.sig.zip ✅
Unzipping firmware... unzipped SMARTDES_020508.fwz ✅
UnGZ firmware... Traceback (most recent call last):
  File "main.py", line 296, in <module>
    c.main()
  File "main.py", line 50, in main
    self.unGZfirmware()
  File "main.py", line 140, in unGZfirmware
    shutil.copyfileobj(f_in, f_out)
  File "/usr/lib/python3.8/shutil.py", line 205, in copyfileobj
    buf = fsrc_read(length)
  File "/usr/lib/python3.8/gzip.py", line 292, in read
    return self._buffer.read(size)
  File "/usr/lib/python3.8/_compression.py", line 68, in readinto
    data = self.read(len(byte_view))
  File "/usr/lib/python3.8/gzip.py", line 479, in read
    if not self._read_gzip_header():
  File "/usr/lib/python3.8/gzip.py", line 427, in _read_gzip_header
    raise BadGzipFile('Not a gzipped file (%r)' % magic)
gzip.BadGzipFile: Not a gzipped file (b'PK')
`

fquinto, 2022-11-24:
Updated script. Can you test it? Thanks!

bubez81, 2022-11-25:
hello, but the problem is the signed firmware isn't it?

bubez81, 2022-11-25:
I want to try but can you add the home touch firmware for a possible choice? 
this is the link:
https://www.homesystems-legrandgroup.com/MatrixENG/liferay/bt_mxLiferayCheckout.jsp?fileFormat=generic&fileName=SMARTDES_020508.fwz&fileId=58107.23188.18516.42630

fquinto, 2026-11-10:
This model is outside what the project supports: nobody has validated a rooted image on it.

For HomeTouch there is a dedicated community project:
[`bticino-hometouch-homekit`](https://github.com/bubez81/bticino-hometouch-homekit).

Closing as out of scope; if someone gets it working, a pull request is welcome.

closed 2026-11-10 by fquinto