fquinto/bticinoClasse300x #3: Elaborate instructions

slyoldfox, 2022-10-26:
This is exciting. I've been looking to have more control over my btcino.

I do have some general questions:

* Could you add how I determine my model number? My screen is cracked since our house was finished, so I can't reach most of the menu's, I am awaiting a new device so I might as well try accessing it.
* My firmware has already been updated to 1.7.19 (includes netatmo camera login support), but I can't login to any account since I can't type numbers and symbols due to the cracked screen.
* Did you manage to find this firmware yet?
* Do you think I can "downgrade" from 1.7.19 to a cracked 1.7.17 ?
* The only software I need is MyHomeSuite is that correct? And a micro-usb cable to connect to the back and my computer?
* The python script modifies the firmware without having to do all manual work?

Veuchez, 2022-11-24:
I approach the request, the guide is not very clear to me.
Does the python script "main.py" modify the firmware without having to do all the manual work described in "Basics steps to get root"?
Thank you.

fquinto, 2022-11-24:
Hi, sorry. Guide is a mix from different authors... and I trying to merge all.

1. mmm is inside some XML file... 
2. :(
3. The repo are updated ;)
4. I don't test it
5. Yes
6. Yes

closed 2022-11-24 by fquinto