fquinto/bticinoClasse300x #18: firmware_versions.md - Community call

lpercetti, 2023-03-30:
Hello,

As the documentation **firmware_versions.md** says about C100X, the version 1.5.4+ sounds not available on the web anymore.
However, people having their firmware cracked already can get the updates locally and can potentially:
- Share the file they have been able to download
- Use tcpdump man-in-the-middle to dump the links where the file is downloaded.

From API man-in-the-middle, I've been able to get following info on my non-hacked device:
`https://api.developer.legrand.com/devicemanagement/api/v2.0/modules/{{deviceId}}/firmware
{
    "deviceId": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "nbOfTry": 0,
    "status": "Pending user consent for installation",
    "currentFirmwareVersion": "1.5.4",
    "id": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "firmwareVersion": "1.5.5",
    "downloadDateTime": 0,
    "installationDateTime": 0,
    "fileSize": 106495633,
    "checksum": "a8fd81f80e07c1d5fbd3e491852123c6",
    "hashAlgorithm": "md5",
    "releaseNote": "# Description : \n- Firmware 1.5.5 for Classe100X16E\n- Minor bug fixes"
}`

So if the hash match, I would be happy to use the fw file of someone else :)

It would help the community! :)

Regards,

lpercetti, 2023-08-20:
Sorry for the late reply.

Yes, you can easily get it using this call:
`GET https://api.developer.legrand.com/servicecatalog/api/v3.0/modules?=`
And a bearer token in the header as authentication.

And then get a list of object in which the "id" key is the deviceId.

lhari84, 2024-02-04:
> Hello,
> 
> As the documentation **firmware_versions.md** says about C100X, the version 1.5.4+ sounds not available on the web anymore. However, people having their firmware cracked already can get the updates locally and can potentially:
> 
> * Share the file they have been able to download

I have found a page offers latest version 1.5.5. But not sure it works for all WiFi-capable Classe 100 models: 

**https://www.homesystems-legrandgroup.com/home/-/productsheets/4221129**

I may try when I have some time left.

fquinto, 2024-02-06:
Firmware links: updated. Thanks.