fquinto/bticinoClasse300x #19: SSH refused -- had to add my MQTT settings in the firmware file before flashing

End0rphi, 2023-04-21:
Hello and thanks for the modified firmware!

If anyone else is having trouble connecting to a modified C300X after flashing:
It is possible to edit the MQTT settings in you local copy of TcpDump2Mqtt.conf before executing main.py.

Does anyone know why I'm getting "Connection refused" with SSH? Do I need to specify my bticinokey?

Thanks!

slyoldfox, 2023-04-21:
try power cycling the unit

I personally run a server.js node server on it so I can access it with "nc" in case that sshd craps out.

msanchezt, 2023-09-02:
> try power cycling the unit
> 
> I personally run a server.js node server on it so I can access it with "nc" in case that sshd craps out.

Sorry for the basic question but... how do you run the JS? If node not available... This is a busybox environment, very limited.

slyoldfox, 2023-09-03:
@msanchezt there are some instructions on https://github.com/slyoldfox/scrypted/tree/main/plugins/bticino on how to put nodejs on your unit.

msanchezt, 2023-09-03:
> @msanchezt there are some instructions on https://github.com/slyoldfox/scrypted/tree/main/plugins/bticino on how to put nodejs on your unit.

That worked: node working now on bticino. Could you recommend a good node server shell-like so I can input commands from nc? Like reboot the sshd, etc...

slyoldfox, 2023-09-03:
> > @msanchezt there are some instructions on https://github.com/slyoldfox/scrypted/tree/main/plugins/bticino on how to put nodejs on your unit.
> 
> That worked: node working now on bticino. Could you recommend a good node server shell-like so I can input commands from nc? Like reboot the sshd, etc...

I use https://gist.github.com/slyoldfox/ec8f8b83d0835807b186e59926980807

msanchezt, 2023-09-03:
> I use https://gist.github.com/slyoldfox/ec8f8b83d0835807b186e59926980807

Just amazing, worked like a charm. Finally, would you mind sharing how do you make sure this runs at startup? A script under /etc/rc5.d/ maybe?

slyoldfox, 2023-09-03:
I added it to /etc/init.d/mosquitto:

in the `start)` function

```
    start)
        echo "Starting Mosquitto message broker" "mosquitto"
	/usr/bin/screen -d -m /home/bticino/cfg/extra/node-v17.9.1-linux-armv7l/bin/node /home/bticino/cfg/extra/c300x-backdoor/server.js
        if start-stop-daemon --start --quiet --oknodo --background  --make-pidfile --pidfile ${PIDFILE} --exec ${DAEMON} -- -c /etc/mosquitto/mosquitto.conf ; then
            exit 0
        fi
        ;;
```

I "installed" the screen program pretty much the same way as node (just added some required libs to /lib)