fquinto/bticinoClasse300x #20: BT-344842

Lockzi, 2023-05-27:
Hello,

I've downloaded and made changes trying to make firmware for C300 EOS work.

Product page: https://www.legrand.pt/diy/classe300eos/

Product sheet page where the firmware download link was found: https://www.homesystems-legrandgroup.com/home/-/productsheets/2987852

**Image of the downloaded firmware extracted**
[image]

**main.py after changes**
`#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""Prepare firmware update."""

__version__ = "0.0.9"

import wget
import zipfile
import tempfile
import shutil
import os
import subprocess
import gzip
import pyminizip

class PrepareFirmware():
    """Firmware prepare class."""

    def __init__(self):
        """First init class."""
        # Last known firmware version for C300X and C100X models
        self.urlC300X = ('https://prodlegrandressourcespkg.blob.core.'
                         'windows.net/binarycontainer/bt_344642_3_0_0-'
                         'c300x_010719_1_7_19.bin')
        self.urlC100X = ('https://www.homesystems-legrandgroup.com/MatrixENG/'
                         'liferay/bt_mxLiferayCheckout.jsp?fileFormat=generic&'
                         'fileName=C100X_010501.fwz&fileId='
                         '58107.23188.46381.34528')
        self.urlMX = ('https://www.homesystems-legrandgroup.com/MatrixENG/'
                         'liferay/bt_mxLiferayCheckout.jsp?fileFormat=generic&'
                         'fileName=MX_020216.fwz&fileId='
                         '58107.23188.56230.57461')

        # Variables
        self.filename = None
        # Contants
        self.password = 'C300X'
        self.password2 = 'C100X'
        self.password3 = 'SMARTDES'
        self.password4 = 'MX'
        self.workingdir = None
        self.prtFrmw = None
        self.useWebFirmware = None
        self.rootPassword = None
        self.SSHcreation = None
        self.removeSig = None
        self.installMQTT = None
        self.notifyNewFirmware = None
        self.mntLoc = '/media/mounted'

    def main(self):
        """Main function."""
        # Get the current working directory
        cwd = os.getcwd()

        # Ask for model: C300X or C100X
        self.model = input('Insert model (C300X, C100X or C300EOS): ')
[... 520 lines cut ...]
App will notify you when a new firmware is available.
Creating temporary folder... created /tmp/bticino-r3cleha9 ✅
Downloading firmware... 
-1 / unknown downloaded MX_020216.fwz ✅
Reading files inside firmware... done ✅
Selecting firmware file... important file is None ✅
Unzipping firmware... Trying to unzip with password: MX ... (please wait arround 95 seconds) unzipped MX_020216.fwz ✅
UnGZ firmware... Traceback (most recent call last):
  File "/usr/src/app/main.py", line 565, in <module>
    c.main()
  File "/usr/src/app/main.py", line 143, in main
    self.unGZfirmware()
  File "/usr/src/app/main.py", line 291, in unGZfirmware
    with gzip.open(f'{self.workingdir}/{self.prtFrmw}', 'rb') as f_in:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/gzip.py", line 58, in open
    binary_file = GzipFile(filename, gz_mode, compresslevel)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/gzip.py", line 174, in __init__
    fileobj = self.myfileobj = builtins.open(filename, mode or 'rb')
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/bticino-r3cleha9/None'
```

Maybe someone could assist? I am thinking it might have to do with that the filenames mention SHA256?

gzone156, 2023-05-29:
The problem is that the C300EOS firmware archive structure is quite different, have a look there:

[image]

We should handle this difference in the script by moving differently across files, but the answer is, before automating the process with the script, have you already tried to hack the firmware successfully? I'm not sure that C300EOS is hackable in the same way as C*00X

Please let us know, otherwise editing the script is just a waste of time.

fquinto, 2026-11-10:
This model is outside what the project supports: its firmware package has a
different structure and nobody has validated a rooted image on it.

Closing as out of scope; if someone gets it working, a pull request is welcome.

closed 2026-11-10 by fquinto