fquinto/bticinoClasse300x #21: BTCINO C100X

gbarrena, 2023-06-17:
I have flashed the C100x with version 1.5.5 modified for mosquito, I have put the mosquito server the ports and the credentials but the intercon does not issue anything to the mosquito broker.

Any ideas?

can anyone help me?

msanchezt, 2023-07-20:
C100X does not have Python3, only Python2.7, you need to modify 2 files under:  /etc/tcpdump2mqtt/, whenever python3 is mentioned replace it by python2

As well, modify file:  /home/root/filter.py

Line #8 with the print should be like this:
     print(line[line.find('*'):]+"\n")

AlbertoAIG, 2023-07-25:
> C100X does not have Python3, only Python2.7, you need to modify 2 files under: /etc/tcpdump2mqtt/, whenever python3 is mentioned replace it by python2
> 
> As well, modify file: /home/root/filter.py
> 
> Line #8 with the print should be like this: print(line[line.find('*'):]+"\n")

Hi,

I can't edit this file, it says read only file.

fquinto, 2024-01-27:
New version of python creator script. This is done there. Thank to @PabloCamp0s

closed 2024-01-27 by fquinto