fquinto/bticinoClasse300x #23: Problem with generate firmware

smarticz, 2023-11-17:
Anyone can send me baked firmware? 

I have problem with 

python3
`ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33'`
or

python

```
import wget
ModuleNotFoundError: No module named 'wget'
```

fquinto, 2024-01-27:
I'm just create a new version. Can you test? First do:

```
sudo python3 -m pip install -r requirements.txt
sudo python3 main.py
```

fquinto, 2026-11-10:
This issue is old and the builder has changed a lot since then (it now lives
in `legacy/`, runs with `docker compose run bticino` or
`sudo python3 legacy/main.py`, and the steps are described in
`legacy/README.md`). Is this still a problem with the current version?

I will close it in two weeks if there is no reply. Please open a new issue
if it comes back.

I cannot share a ready-made firmware file: the project never distributes
BTicino images, only the tools and the checksums of the official ones.