fquinto/bticinoClasse300x #27: What works and what not

unecc, 2024-01-26:
Hi to all, i've 300x and homeassistant, bticino app is a shit an i don't receive notification when seomeone rings to me.
What can i do with your code? What works and what not?
Where can i find this informations?

Thanks a lot of work of the developer, if i like it i will make donation for go on. Bye

fquinto, 2024-01-27:
I'm working on it, I like to do things easy and this is hard work. Finally I like to perfect integration with HA.

For donations: your are welcome thanks!!

https://www.paypal.com/donate/?business=XQQJHGVPHLD7W&no_recurring=0&item_name=Thank+you+for+collaborating+and+helping+the+development+of+software+improvements.&currency_code=EUR

lhari84, 2024-02-05:
@fquinto 
As far as I understand current state of the project, it's possible to get states (camera on/off,...) and do actions (open door,...) in/from home assistant. 
On the other hand, what I'm missing is to open the video stream in home assistant too. Do you see a chance this will be possible too? 
Thanks for your work, I will try out the firmware modification when I find some time and if it turns out to be useful for me (as I'm currently on HomeKit + Homebridge instead of Home Assistant and have to check, if I can pass things through from there), I will also do a donation. 

One more thing, if you haven't seen it in the other issue. I guess I have found download for latest firmware version of Classe 100x. Maybe it helps to include v 1.5.5 in you script. 
https://www.homesystems-legrandgroup.com/home/-/productsheets/4221129

fquinto, 2024-02-06:
It will be possible in "easy way" in the future to see open video stream. Not yet. But some people get a solution. I will work to integrate all needs together, but my free time are limited.
Another task is to organize the "readme" file.
Firmware links: updated. Thanks.

filou01, 2024-02-09:
Does this firmware only extend the original one with mqtt capabilities or does it also disable some online services like the ios app connection or the recording functionality?
Thanks

fquinto, 2024-02-09:
It is only to be root on the device.  He is not ready, from my point of
view, for anything else.  I also do not recommend activating MQTT.  For the
moment, nothing more.

filou01, 2024-02-11:
Thank you. 
I dont understand why you don‘t recomment to enable mqtt. As far as i understood the information provided on the readme page of this project the mqtt extension is the main goal of this github project. Can you please further explain me what are the advatages to use your firmware compared to the original one? My goal is to integrate the doorbell into Home Assistant. 
Thank you and cheers
Christoph

weltall, 2024-02-13:
> Thank you. I dont understand why you don‘t recomment to enable mqtt. As far as i understood the information provided on the readme page of this project the mqtt extension is the main goal of this [...]

From what i've seen you've a problem with #25 if you try to generate a firmware with mqtt and if you didn't set wifi before the flash you've a possibly unrecoverable brick. Personally i did a firmware without mqtt and integrated it manually through ssh by comparing the two images, still not running it automatically tough.

fquinto, 2026-11-10:
The README has been rewritten to answer exactly this: a "Model status" table
(what is validated on which model and firmware), a table of every software
piece with its status, and a "Known problems" section. The bridge has its own
status table in `bticino_bridge/README.md`.

Short version for a Classe 300X on firmware 1.7.x: doorbell events, door
opening, stair light, answering machine, live video during a call and
snapshots work in Home Assistant. Doorbell notifications no longer depend on
the vendor app.

Closing; if something in those tables is unclear or wrong, please say so.

closed 2026-11-10 by fquinto