fquinto/bticinoClasse300x #29: Support for C100X firmware v2.0.12

ctmayhew, 2024-02-15:
Do you think this would work with version 2.0.12 of the firmware for C100X16E available to download here?
https://www.homesystems-legrandgroup.com/home/-/productsheets/2595814

This is the non-Door Entry App version of the firmware which was included on all units purchased from mid 2023 onwards. 

Thanks !

ctmayhew, 2024-02-28:
Just sent a donation via PayPal. Thanks for your work on this project !

fquinto, 2024-02-28:
Hello! Thanks for your support!! I just received your donation!!
I cannot assure you that this version is suitable for use on your C100X. In particular, I do not have the C100X version (I have the older C300X). For the moment I suggest waiting a little longer and securing to avoid possible terminal blockages. There have been some cases of terminal blocking (brick).

maximushugus, 2024-03-26:
I can confirm that my C100X H+S came preinstalled with firmware v2.0.12 yesterday. For the moment I'll wait until a confirmation that I can use this mod.
Thanks for your work !

fastcolors, 2024-04-28:
I got the same firmware, I see you have the link  for it in firmware_versions.md.

would it work if i modify main.py in order to use that link ?

fquinto, 2024-05-05:
Try! and cross fingers!

msanchezt, 2024-07-07:
Same problem here. Got C100X with firmware v2.0 :/

saberraz, 2024-11-22:
I would really appreciate if you could add support for firmware 2.x. I have[ this model ](https://www.homesystems-legrandgroup.com/it/home/-/productsheets/2595814) and there is already a firmware version 2.x available on that page and also linked in [firmware versions](https://github.com/fquinto/bticinoClasse300x/blob/main/firmware_versions.md). 

ps: I had already made a donation :)

fquinto, 2024-11-23:
I really appreciate your donation but I only have the old version C300, not
the new C100. Without the new one it is very very difficult to make any
progress on my part.

marcob79, 2024-11-23:
Supposing to have a c100x with firmware 2.x and ssh already enable, it can be useful to retrieve some information to implement the unlock procedure for 2.x?

fquinto, 2026-11-10:
Units on firmware 2.x (the ones that pair with the **Home + Security** app)
are a different product line: different software, Netatmo cloud. This
project only applies to the firmware 1.x line (the units that pair with
**BTicino Door Entry**), and a 2.x unit cannot be downgraded with it — the
firmware packages are not interchangeable.

For 2.x units the working option today is the cloud integration
[`bticino_intercom`](https://github.com/k-the-hidden-hero/bticino_intercom),
which needs no firmware modification. The README now has a section "Is this
for my unit?" explaining how to tell the two lines apart.

Closing as out of scope. Thanks for asking.

closed 2026-11-10 by fquinto