fquinto/bticinoClasse300x #31: Support for Bticino Classe 100 X16E (Model 344932)

afrixx, 2024-04-21:
Hi,
thanks for providing this great software project!

There seem to be different models called Classe 100 X16E.
My one has the model number 344932
This number I cannot select in the "My Home Suite" software.

The model including firmware file can be found at the official webpage:
https://www.homesystems-legrandgroup.com/home/-/productsheets/2595895

Is there any way to import this model into the "My Home Suite" software?

Thanks for any help and support!
Andreas

sstidl, 2024-05-31:
Same here. Didn't get a connection with "My Home Suite"

sstidl, 2024-06-02:
Today I took the risk and tried the instructions with the firmware from the page above and was successful.

I downloaded the file from there, named it c100x_.fwz and put it into original_firmware/ folder.
then did the customization with the python script.

in myhome_suite i selected 344782 (because it was the highest number for a Classe 100X16E).

Important step here is to select firmware version 1.4 (not the preselected 2.0)

[image]

then contiued as in the instructions (select firmware file, upload firmware)

important: connect the usb cable AFTER pressing the "firmware upgrade" button

sstidl, 2024-06-03:
Today the device got an update to 1.6.5 and I am locked out again.

Also the link above changed to version 02.00.12. 
So be careful not to delete the old version.

Don't have time to try to revert at the moment.

afrixx, 2024-06-03:
Oh no!
Good to know, how you solved the topic - but where does this 1.6.5 firmware come from?
Another model?

sstidl, 2024-06-03:
> Oh no! Good to know, how you solved the topic - but where does this 1.6.5 firmware come from? Another model?

It self updated! So it seems...

sstidl, 2024-06-03:
Ok sorry my mistake. The device is still 1.5.5. The android app is 1.6.5.

sstidl, 2024-06-05:
After a power cycle it is now reachable. It seems that the dropbear sshd crashed.

sstidl, 2024-08-01:
> The model including firmware file can be found at the official webpage:
> https://www.homesystems-legrandgroup.com/home/-/productsheets/2595895

the link has changed. it's now 
https://www.homesystems-legrandgroup.com/en/home/-/productsheets/4221129

the download-link of version 1.5.5 firmware is 
https://www.homesystems-legrandgroup.com/MatrixENG/liferay/bt_mxLiferayCheckout.jsp?fileFormat=generic&fileName=C100X_010505.fwz&fileId=58107.23188.62332.48840

jcshaeffer, 2024-11-28:
Having a Classe 100 X16E  model  344932 too, i wonder:

1. What is the way back, if something fails? Is it possible to download the existing firmware via USB as a backup first?

2. Currently there is the version 02.02.02 on the device. The lastest version i can download is still 02.00.12. is it a problem going back to an older version? Does anyone have experience with the 2.xx version?

3. Sorry if it is a stupid question, but: is there an "easier" alternative way to run and open the way to an ssh server? For, than, performing all the steps in a terminal session?

Would be great getting access to this device. It's really annoying how closed it is, while all the technology is there...

tamet83, 2024-11-28:
Hi,

It's better not downgrade the firmware on new devices because of disconnection issues.
I contacted directly Bticino in Italy to solve this issue and spoke with a technician.
They pushed 3 updates to solve the issue.

The latest firmware version is 2.03.01

jcshaeffer, 2024-11-28:
Thanks for answering, Matteo.

I also received the latest firmware version 2.03.01. BTW, is it possible to stop the device updating automatically and noticing about a possible update instead? (I did not find this option.)

So I am not able using the Fran's solution? That's a pitty.

I do not exactely understand "They pushed 3 updates to solve the issue.": to solve which issue exactely?

tamet83, 2024-11-28:
When I installed the device i had the firmware 2.01 or something like that. I noticed frequent disconnection of the device from my wifi so I called Bticino fabric here in Varese where I live and spoke with a technician. He contacted me several times because they couldn't find the causes of the issues. They had to made three new firmware versions to solve it. For what I understood they push the firmware to devices to upgrade it. It's not something you can do by yourself.

Anyway unfortunately seems that we will never be able to implement Fran's solution on our devices.

Pete1612, 2024-12-22:
Stupid question but has anybody ever tried to flash the V2.X onto a V1.5 device such as the Model 344932 to have it work with the Home + Security app?
I suspect that they just took to the same hardware for V2.X as they did on V1.5….

slyoldfox, 2024-12-22:
> Stupid question but has anybody ever tried to flash the V2.X onto a V1.5 device such as the Model 344932 to have it work with the Home + Security app?
> 
> I suspect that they just took to the same hardware for V2.X as they did on V1.5….

As far as I heard it's different hardware... don't

fquinto, 2026-11-10:
The legacy builder has a Classe 100X option (firmware 1.5.x), contributed by
the community; it has not been validated by me on a real 100X. For 100X
units on firmware 1.x there are also dedicated projects, listed under
"Related projects" in the README. On the question above about flashing 2.x
onto a 1.5 unit: as far as is known the two lines are not interchangeable.

Closing this one since it has turned into a general 100X thread; please open
a new issue for a specific problem.

closed 2026-11-10 by fquinto