fquinto/bticinoClasse300x #32: Information about firmware 1.5.5

Rescok, 2024-06-10:
Hi Fquinto,

thank's for your job!! I would like to ask just one question because I have the C100X model with firmware 1.5.5...
I see that firmware is available online and you have put the correct download link in the "firmware_version.md" file, however in "main.py" there is no possibility to insert 1.5.5 in the automatic script to generate the custom firmware (error wrong firmware appear on sceen). Is there a reason for excluding this firmware from the automatic script?

sstidl, 2024-07-29:
Try this 
https://github.com/fquinto/bticinoClasse300x/issues/31#issuecomment-2143834566

fquinto, 2026-11-10:
Support for 1.5.5 was added to the builder later (pull request #33).

This issue is old and the builder has changed a lot since then (it now lives
in `legacy/`, runs with `docker compose run bticino` or
`sudo python3 legacy/main.py`, and the steps are described in
`legacy/README.md`). Is this still a problem with the current version?

I will close it in two weeks if there is no reply. Please open a new issue
if it comes back.