fquinto/bticinoClasse300x #35: C100X firmware 1.5.7

sstidl, 2024-07-29:
My device says there is 1.5.7 available.
Will try to update it next week.

I copied the file from /home/bticino/cfg/extra/FW/UPDATE.fwz
and will try to use it just like here
https://github.com/fquinto/bticinoClasse300x/issues/31#issuecomment-2143834566

sstidl, 2024-08-02:
I found the URL which is used by the device to download 1.5.7:
https://eliotdmstrg.blob.core.windows.net/bs-classe100x/C100X_010507.tar.gz
In the tar file there is a .bin file which is the same as the .fwz files.
I checked md5sum from this file and the Upgrade.fwz directly on the device (as stated in Readme)

fcaronte, 2024-08-24:
i can confirm this firmware is working on c100x

sstidl, 2024-11-01:
Now the version 1.5.7 is officially for download at
https://www.homesystems-legrandgroup.com/en/home/-/productsheets/4221129

closed 2025-05-06 by fquinto