fquinto/bticinoClasse300x #36: Cannot connect via USB to C100X16E

dirz88, 2024-07-31:
Hi all
I've builded the firmware and failing to flash it to my device. The software MyHOME simply doesn't recognize it. It fails with the error "Unable to contact device". 

I've also tried it like suggested here: https://github.com/fquinto/bticinoClasse300x/issues/31#issuecomment-2143834566

Could anyone give me a hint to solve this?
Thanks a lot.

sstidl, 2024-08-01:
Did you connect the USB cable after starting to flash?

I did use a fresh install of win11 without any additional security software. It didn't work on a laptop with trendmicro antivirus installed.

dirz88, 2024-08-01:
Yep - I‘ve connected it after starting to flash. I‘ve tried 3 notebooks with WIN10, WIN11 with fully disabled Antivirus. I also rebooted the Classe by unplug the BUS, unfortunately that didn‘t help.

I‘ve also tried 3 different USB cables. According to the iOS App ‚Securty‘ the Classe does have version 2.03.01 installed. I‘ll have a look if I can try to get a PC with plain OS installed. Thanks for helping so far.

sstidl, 2024-08-01:
I don't know if a device running 2.x will work. The project here only supports 1.x firmware devices. Never heard of anyone downgrading so far.

tarbax, 2024-12-03:
Any succes with firmware version 2.03.01 ? 
I also have that firmware...

sstidl, 2024-12-03:
As stated in other discussions, firmware 2.x is not supported by this repo

majorhector, 2025-06-07:
If you have 2.x firmware you can use btcino_intercom integration in HACS without flashing anything :)

https://github.com/k-the-hidden-hero/bticino_intercom

fquinto, 2026-11-10:
Units on firmware 2.x (the ones that pair with the **Home + Security** app)
are a different product line: different software, Netatmo cloud. This
project only applies to the firmware 1.x line (the units that pair with
**BTicino Door Entry**), and a 2.x unit cannot be downgraded with it — the
firmware packages are not interchangeable.

For 2.x units the working option today is the cloud integration
[`bticino_intercom`](https://github.com/k-the-hidden-hero/bticino_intercom),
which needs no firmware modification. The README now has a section "Is this
for my unit?" explaining how to tell the two lines apart.

Closing as out of scope. Thanks for asking.

closed 2026-11-10 by fquinto