fquinto/bticinoClasse300x #37: Unable to make it work 300x (home automation)

Christofke, 2024-09-21:
I was able to create the "updated" software for the controller without any error messages (that is if sig is removed and MQTT is removed). I was also able to upload the mod files to the controller, and that is it. Connection ssh gives me "connection refused". I really would love to be able to have my gate automated through home-assistant or homekit. I have checked the repository @fquinto and @slyoldfox but I can not make them work and need some help. I'm not a programmer, so only basic knowledge. What the modified software is concerned I'm using Ubuntu on a windows PC. I tried also on a raspberryPi 4, but could not complete the repo without errors.

slyoldfox, 2024-09-21:
Check the Telegram channel for lots of info and help from members

Christofke, 2024-09-22:
Would it be possible to send me a working modified new 300x-xxxx…fwz file. I have the impression that this is where my system fails.
Thanking you in advance.
Ps: i did check the telegram channel, but the variety of errors discussed is overwhelming and often too complex for me to understand. I believe a step by step manual is what is required.

fquinto, 2026-11-10:
This issue is old and the builder has changed a lot since then (it now lives
in `legacy/`, runs with `docker compose run bticino` or
`sudo python3 legacy/main.py`, and the steps are described in
`legacy/README.md`). Is this still a problem with the current version?

I will close it in two weeks if there is no reply. Please open a new issue
if it comes back.

I cannot share a ready-made firmware file: the project never distributes
BTicino images, only the tools and the checksums of the official ones.