fquinto/bticinoClasse300x #38: /etc/tcpdump2mqtt not existing

adicorra, 2024-12-08:
Hello,

I Managed to flash the firmware on C300X v1.7.19. No errors and MQTT installed.

Unfortunately I cannot see the /etc/tcpdump2mqtt   directory

What can be wrong?

Thanks

frankvoelker, 2025-07-16:
Same here. No /etc/tcpdump2mqtt folder !!!

Any news on this?

frankvoelker, 2025-11-23:
I created new firmware with `MQTT_HOST` set in mqtt_scripts/TcpDump2Mqtt.conf but same problem. I can SSH into the machine but there is no `/etc/tcpdump2mqtt`

any ideas?

frankvoelker, 2025-12-02:
> If you check my last commit, please, provide feedback. Thanks. [dc60716](https://github.com/fquinto/bticinoClasse300x/commit/dc6071638a566e3cee392024683b3480668bbc79)

also I noticed that you have to select the "install mqtt server" during the setup. this should be written in the readme/guide ^^if you don't there will also be no /etc/tcpdump2mqtt

fquinto, 2026-11-10:
This issue is old and the builder has changed a lot since then (it now lives
in `legacy/`, runs with `docker compose run bticino` or
`sudo python3 legacy/main.py`, and the steps are described in
`legacy/README.md`). Is this still a problem with the current version?

I will close it in two weeks if there is no reply. Please open a new issue
if it comes back.