fquinto/bticinoClasse300x #39: [Question] BTicino 344652

abonforti, 2025-01-21:
I understand this is probably not the right place to ask, but worth to do anyway, in case, just close the issue.
I own a non-wifi version model 344652. Is there any BTicino (or third party, perhaps diy ESP) accessory that can be plugged into this model to make wifi-compatible and, hence, use this project?

slyoldfox, 2025-01-21:
I am not a bticino engineer so I don't know anything about the unit.

If it has a usb on the back and you manage to connect to it with the MyHome software, it probably means it has a network enabled usb port.

On C300X units, it uses `usb_f_rndis` to establish ethernet over usb communication. So this means there's a network layer on top of the usb0 connection.

Either way it starts by having a look if you can access it with MyHome (it will install all the required TCP RNDIS drivers for you if you're on windows). And then having a look if you can ping it, if so you might be able to bridge that device and you'd have some sort of wired ethernet connection (don't ask me how stable it is).

fquinto, 2026-11-10:
As explained above, this project needs a unit with network access (the
Classe 300X / 100X connect over Wi-Fi). I do not know of an accessory that
adds that to the 344652. Closing as answered.

closed 2026-11-10 by fquinto