fquinto/bticinoClasse300x #40: Any way to get a "screenshot" of an incoming video call?

Alloc86, 2025-03-13:
Hi,

I was wondering if it's somehow possible to capture like the first frame (or maybe one a few frames later as the first frame might not be a good image yet) when someone uses the door bell? I.e. a copy of a frame from the video stream that is already shown at that time on the internal door station?

Would love to be able to automatically send a frame from the stream when someone uses the door bell e.g. in base64 through MQTT or another medium so I can associate an image with the event that someone was using the bell.

Cheers,
Chris

PS: Good job on this stuff!

KSuondProject, 2025-07-31:
> Hi, I was looking for exactly the same thing.
> 
> I found a solution that works well for my use case.
> 
> Requirements:
> 
> * The voicemail/aswm feature must be enabled
> * ffmpeg installed on the C300X device
>   Just ffmpeg: [github.com/slyoldfox/ffmpeg-for-bticino](https://github.com/slyoldfox/ffmpeg-for-bticino)
>   Or with additional features: [github.com/slyoldfox/c300x-controller](https://github.com/slyoldfox/c300x-controller)
> 
> When the doorbell button is pressed (TX topic with payload: _8_1#1#4#21*10##), a script is triggered that extracts a frame and sends a base64-encoded image to the MQTT broker (to the snapshot topic).
> 
> There might be other or even better ways to achieve this, but this setup works perfectly for me.
> 
> Script:
> 
> #!/bin/sh
> 
> RAW_FILE="/tmp/aswm.raw"
> FFMPEG="/home/bticino/cfg/extra/c300x-controller/ffmpeg"
> MQTT_CMD="/usr/bin/mosquitto_pub"
> MQTT_HOST="192.168.1.1"
> MQTT_PORT="1883"
> MQTT_USER="user"
> MQTT_PASS="pass"
> MQTT_TOPIC="bticino/snapshot"
> MAX_RETRIES=20
> RETRY_DELAY=1  # Seconds between retries
> 
> STABLE_DELAY=1  # Seconds to wait and check file stability
> 
> for i in $(seq 1 $MAX_RETRIES); do
>     if [ -f "$RAW_FILE" ]; then
>         SIZE1=$(stat -c %s "$RAW_FILE")
>         sleep $STABLE_DELAY
>         SIZE2=$(stat -c %s "$RAW_FILE")
> 
>         if [ "$SIZE1" -eq "$SIZE2" ]; then
>             echo "[OK] File is stable. Extracting and sending image..."
> 
>             $FFMPEG -y \
>                 -f rawvideo -pix_fmt yuv420p -s 688x480 \
>                 -i "$RAW_FILE" \
>                 -frames:v 1 \
>                 -q:v 5 \
>                 -vf "select=eq(n\,0),scale=640:360" \
>                 -f image2pipe - \
>             | base64 \
>             | /usr/bin/mosquitto_pub -h "$MQTT_HOST" -p "$MQTT_PORT" -u "$MQTT_USER" -P "$MQTT_PASS" -t "$MQTT_TOPIC" -s
>             "[DONE] Image sent successfully."
>             exit 0
>         else
>             echo "[WAIT] File size still changing... ($i/$MAX_RETRIES)"
>         fi
>     else
>         echo "[WAIT] File not found or empty... ($i/$MAX_RETRIES)"
>     fi
>     sleep $RETRY_DELAY
> done
> 
> echo "[FAIL] Max retries reached. No image was sent."
> exit 1

Hi Jojo!

Do you know if your script could work also on C100X?

Thanks!

KSuondProject, 2025-08-02:
> Sorry, I have no idea. You can check if a raw file is created and if ffmpeg is working on the c100x.

Thank you anyway, maybe I'll try to check if it could works.

jojo18222, 2025-10-04:
You can use the Rx topic to execute commands:
'{"command": "execute_command", "file_path": "", "data": "/home/bticino/cfg/extra/c300x-controller/snapshot_aswm.sh &"}'

or direct via ssh in your script.

fquinto, 2026-11-10:
This is now built into the bridge: when the doorbell rings it takes a
snapshot from the entrance-panel camera and publishes it over MQTT as a
Home Assistant camera entity ("doorbell camera"), so you can attach it to a
notification. Live video during the call is also available over RTSP.
Details in `bticino_bridge/docs/HOME_ASSISTANT.md` (sections "Doorbell
camera" and "Live video during a call").

The shell-script approach posted above still works for the script-based
integration. Closing; reopen if the snapshot does not work on your unit.

closed 2026-11-10 by fquinto