fquinto/bticinoClasse300x #44: Firmware auto-updates to 1.5.8 despite manual installation of 1.5.7 C100X

lpozzo90, 2025-08-04:
Hi,
while preparing the firmware using GNU/Linux as described in this repository, I selected **NO** when main.py prompted **"Received notification for new firmware?"**.

The firmware preparation completed successfully, and version **1.5.7** was created. I then installed this version on the Intercom.

Initially, the Intercom worked correctly and accepted SSH commands. However, after a few minutes, it automatically updated itself to version 1.5.8, overwriting the manually installed version.

To investigate, I tried uninstalling the mobile app to see if the update information was coming from there, but the Intercom still updated itself automatically.

Is there a way to prevent the Intercom from auto-updating the firmware after manual installation?

Thanks in advance for your help!

fcaronte, 2025-08-05:
As for the 1.5.7 seem the link to get the firmware work changing last numero to 8, we need to try if mod work for new versions too

https://eliotdmstrg.blob.core.windows.net/bs-classe100x/C100X_010508.tar.gz

fcaronte, 2025-08-05:
I don't think is possible, but maybe I'm wrong, I don't know if exist a way without flash from zero

beeguy1234, 2025-08-09:
My C100x auto-updated to 1.5.8 so I lost my connection, so I had to install all over.
Which domain(s) do I need to block to prevent it receiving updates again ? 
(Or can I block by default and only whitelist domains needed for app functionality?)
I point to my pi-hole via /etc/resolv.conf

blka, 2025-08-09:
@beeguy1234 probably the same from where firmware is downloaded. It has been listed above.

sstidl, 2025-08-10:
> [@beeguy1234](https://github.com/beeguy1234) probably the same from where firmware is downloaded. It has been listed above.

I dont think so. Last time it was a cloud url. See my contribution here:
https://github.com/fquinto/bticinoClasse300x/issues/35#issuecomment-2264923223

Drums7, 2025-12-12:
> > `Is there a way to prevent the Intercom from auto-updating the firmware after manual installation?` - i think it's not related to manual installation, as my bticino c100x has been updated automatically and i think there is no way to disable autoupdate. maybe you could overwrite the domain on your local dns (after changing hardcoded google and cloudflare dnses in `/etc/resolv.conf`)
> 
> I try to mod the new version and script finished without error. I see it ask if you want to enable ota update notifications and default is yes, maybe if set to no you avoid this self update. Actually I'm not at home to test if new modded version is working but if work I can put here the new modded firmware just ready to be installed

"Hi, can you publish the latest version? I have 1.5.8, but shortly after installing it I lost access to the 100x. I think the firmware gets overwritten with the same version or my file is corrupted. Please send it so I can try it."

fquinto, 2026-11-10:
The update is pushed from the vendor cloud, so the builder option alone does
not prevent it, as others noted above.

There is work in progress on a "cloud-survival" mode that neutralises the
on-device updater in a reversible way. It has only been tried on a Classe
300X (firmware 1.7.x) and is not validated on a 100X, so I do not recommend
relying on it yet. I will update this issue when there is something you can
safely test on a 100X.